frp-arduino/frp-arduino #1: Problem: Same output can be assigned to twice

rickardlindberg, 2015-01-13:
This creates programs that does not make sense.

rickardlindberg, 2015-01-14:
The following code compiles but the resulting behavior is that two different streams write different values to the same pin:

```haskell
main = compileProgram $ do

    toggled <- def $ clock ~> toggle

    pin13 =: toggled

    pin13 =: toggled ~> invert
```

Preferably, the DSL prevents such usage.

mvcisback, 2015-01-15:
Maybe the `DAGState` should have a list of identifiers rather than a counter. That way you can check uniqueness.

I.E.

```haskell
data DAGState = DAGState
    { ids :: [Identifier]
    , dag       :: DAG.Streams
    }
```

mvcisback, 2015-01-15:
Presumably, then the check would happen in `(>>=)`

mvcisback, 2015-01-15:
In addition to disallowing multiple `(:=)` the DSL should also make sure the input and output pin sets are disjoint. That might be done by partitioning ids into 2 sets like:

```
data DAGState = DAGState
data DAGState = DAGState
    { inIDs :: [Identifier]
    , outIDs :: [Identifier]
    , dag :: DAG.Streams
    }
```

and asserting `intersect (inIDs s) (outIDs s) == []`

rickardlindberg, 2015-02-06:
The comment about input and output sets should also be solved by #36.

PR #36 (2015-02-06): Update examples in doc + Fix same output can be assigned to twice
This should close #1.
closed 2015-02-06 by mvcisback