frp-arduino/frp-arduino #6: Reforumlate Arduino Controller as Hybrid I/O Automata (HIOA)

mvcisback, 2015-01-13:
Currently, the generated Arduino code seems to act like a single state machine where the dynamics are governed by the Stream binding to pins (which can be seen as the variables)

I propose introducing `Events` (which correspond to guards in HIOA) for transitioning between states.

These could look like

[image]

or perhaps

[image]

The syntax for the two state system could look like:
```haskell
type Behavior a = Stream a

switcher :: Behavior a -> Event (Behavior a) -> Behavior a

s1 :: Behavior Int
s2 :: Behavior Int

s3 :: Behavior Bool
s3 = isEven `mapS` s1

s4 :: Behavior (Int, Bool)
s4 = s3 <> s1

s5 :: Behavior (Bool, Int)
s5 = s3 <> s2

-- Triggered if First element is True, Holds behavior to be used if Triggered
buttonEvent :: Behavior (Bool, a) -> Event (Behavior (Bool, a))

e :: Event (Behavior (Bool, Int))
e = buttonEvent s5

pin1 := switch s4 e
``` 
Which would change from behavior `s4` to behavior `s5` when `s1` is even

rickardlindberg, 2015-01-14:
Programs currently undergo the following transformation:

    DSL -> DAG -> C-code

We write using constructs in DSL. Those commands will build up a DAG data structure. Some of the information that is present in the DSL is not present in the DAG. From the DAG we generate C code.

From your examples, it looks like only the DSL need to change. The DAG data structure (and hence the C-code part) can stay the same. Have I understood correctly?

On the other hand,

> I propose introducing Events (which correspond to guards in HIOA) for transitioning between states.

sounds like you propose a change to the structure of the generated C code.

mvcisback, 2015-01-14:
> sounds like you propose a change to the structure of the generated C code.

Yes, as far as I could tell looking at the generated code, it can't easily handle transitions between behaviors based on some external event.

Thus, after each clock step, (loop iteration), perhaps we should see if the state of the system needs changing. Here is some psuedo code for the process:

```python
while True:
    input_vals = read(input pins)
    output_pins = write(output_vals)
    out_vals, state_vals = evolve_state(input_vals, out_vals, state_vals, time)
    event = check_events(input_vals, state_vals, time)
    if event is not None:
        state_id, evolve_state, check_events = next_state[state_id]
```

The big difference is the state/event dispatch

The model I have in mind is from a [paper](https://github.com/mvcisback/line_follower/blob/master/path_follower.pdf?raw=true) describing a line following robot

[image]

In this case, based on the sensor (HIT/MISS) which could be a `Stream Bool` I want to change how the forcing `F_r, F_l` acts. This depends on the state of the system.

rickardlindberg, 2015-01-15:
I've worked on the code generator today. Is is more readable now. I might have time to push the changes tonight. But it needs some more work.

The improved code generator should be easier to experiment with.

mvcisback, 2015-01-16:
Just watched the strange loop talk.

Indeed, in some sense Event based switching requires a "dynamic graph". That said, for the frp-arduino use case a lot of the benefits he mentioned either not applicable or unsubstantiated (I just added his thesis paper to my reading queue, so I'll see if he elaborates).

# Benefits of Static Signal Graph:
## Hotswapping: 
- Because the code is flashed onto the device, I'm not sure how this would work. 
- In theory if the device were plugged in via usb, one could use the standard firmata instead or implement a simulator.
- Alternatively, I suppose the starting state could be compiled in, but implementing this kind of a thing seems non-trivial.
- That said, his reasoning why this is a benefit of the static signal graph is suspect, particularly in our case sense signals sources and sinks are bound to pins which are finite (and presumably known).

## Time Travel Debugger:
- Same as Hotswapping

## Rigid Structure
- This is perhaps an actual benefit. That said, I wasn't really convinced by his example of the mouse signal. To implement the about HIOA, I would need to capture the current state in the data type. Specifically something like:

```haskell
data State = {
     var1 :: A
     var2 :: B
     var3 :: C
     loc :: Identifier -- used to decide how to evolve state
} 
```

The problem is now I have to couple my behaviors to the structure of `State`. In my opinion, the `loc` is a property of the DSL, and should be encoded directly into it via a `Event` and `switcher`.

# Relation to HIOA

I think that modeling the program as a declarative specification for a HIOA resolves a number of issue presented. What he said about Automata in Arrowized FRP more or less solidified the notion that this could be done.

rickardlindberg, 2015-01-16:
You present a lot of interesting material that I have to read up on. I came across FRP via Elm a few years ago and thought it might be a good paradigm for programming the Arduino. I don't really have a strong theoretical background in FRP and reading papers is a bit difficult for me.

I appreciate having you involved in this project :)

mvcisback, 2015-01-20:
So, after thinking about it/trying it. `Stream a` can't be encoded by `Monoid` or `SemiGroup` because 
`(<>) :: m a -> m a -> m a` not `(<>) :: m a -> m b -> m (a,b)`. That's not to say it isn't either of those semantically, it just the type `Stream a =\= Stream b =\= Stream (a,b)`

mvcisback, 2015-01-20:
incidentally, [Data.ByteString](http://hackage.haskell.org/package/bytestring-0.10.4.1/docs/Data-ByteString-Builder-Prim.html#v:-62--42--60-) features an interesting function (found via hoogle) 
`(>*<) :: Monoidal f => f a -> f b -> f (a, b) infixr 5`

which does a very similar thing.... although I don't see any other uses of this

mvcisback, 2015-02-02:
@rickardlindberg So I'm going to be doing some research into this soon. I'll post any preliminary notes on this. In the mean time I'm going to try refactoring to allow for multiple backends

rickardlindberg, 2015-02-02:
Sounds good to me.

rickardlindberg, 2015-02-04:
The simplest solution that I can think of to support multiple backends is to parse arguments here:

https://github.com/frp-arduino/frp-arduino/blob/master/src/Arduino/DSL.hs#L144

If for example --backend2 is found in command line arguments, we can call another function to generate the C file.

This does not require any refactoring.

mvcisback, 2015-02-06:
So, after looking around. There _is_ infact some (indirect) research on this.

Specifically, there has been work on showing that [linear-time temporal logic can serve as a proof of an FRP program](http://ect.bell-labs.com/who/ajeffrey/papers/plpv12.pdf).

This means there's a Curry-Howard like correspondence between the 2.

More over, there is a [well known algorithm](https://en.wikipedia.org/wiki/Linear_temporal_logic_to_B%C3%BCchi_automaton) to transform a linear temporal logic formula into a Buchi Automata (which is a special case of Hybrid I/O Automata).

Lastly, Buchi Automata are naturally formulated in procedural languages like C (which is what we compile to).

mvcisback, 2015-02-25:
@frp-arduino/owners hey, figured I'd check in.

So I've now spent some time learning agda, with the hopes of providing stronger type guarantees on the front end.

Probably not surprising given what I said previously, but someone built an agda library for FRP using Linear Temporal Logic (LTL) as the basis for the types. https://github.com/agda/agda-frp-ltl/

They, _also_ implemented an ecmascript backend https://github.com/agda/agda-frp-js.

I plan to play with the abstraction, and see how well it could serve as a front end. If that works out, I plan to write a generic reduction to a Hybrid I/O Automata which is much easier to directly translate to C or Assembly.

rickardlindberg, 2015-02-25:
Cool. Good luck :-)