frp-arduino/frp-arduino #11: C code generation using Ivory or Copilot

mvcisback, 2015-01-16:
So I've been looking at the C code generation, it reminds me a lot of 2 projects from Galois that we might consider leveraging (although that can sometime be painful in Haskell).

The first is [Ivory](http://ivorylang.org/ivory-concepts.html) which is EDSL for systems programming.

The second is [CoPilot](https://github.com/leepike/Copilot) which came before Ivory. This actually almost exactly matches the current Stream transformations.

rickardlindberg, 2015-01-16:
Interesting. I'll have to read more about them.

rickardlindberg, 2015-01-17:
I guess we can have multiple backends and experiment with. Then perhaps introduce a switch to choose backend from the command line.

mvcisback, 2015-01-17:
@rickardlindberg I think that's a really good idea. I have seen many great Haskell libraries do something similar. Specifically, [diagrams](http://projects.haskell.org/diagrams/) does this well using

```haskell
import Diagrams.Prelude
import Diagrams.Backend.SVG.CmdLine

main = mainWith (circle 1 :: Diagram B R2)
```

Where `B` is an alias for the backend

my suspicion is that we could add a type parameter for `Action`, i.e. `Action b a`, where `b` is the backend and `a` is still the `Action` type.

Thus `compileProgram` becomes `compileProgram :: Action B a -> IO ()` where `B` is exported based on the backend.

rickardlindberg, 2015-01-21:
The title sounds more like a solution to me. But what is the problem?

mvcisback, 2015-01-21:
@rickardlindberg I imagine the problem is that their exist C generation libraries that have solved many of this issues (such as making sure literals aren't referenced by address) already.

rickardlindberg, 2015-01-21:
So if using a library for code generation makes our codebase smaller and easier to understand, then I think that's a problem worth solving.

joeyh, 2020-01-25:
I've built something not unlike frp-arduino on top of copilot http://joeyh.name/blog/entry/announcing_arduino-copilot/