frp-arduino/frp-arduino #13: What is your interest in frp-arduino?

rickardlindberg, 2015-01-16:
Mine is mostly to experiment and learn about designing and implementing a useful DSL.

Then maybe I will do some Arduino project using the DSL :)

rickardlindberg, 2015-01-17:
I'm also interested in trying to build a community around this project. Inspired by [ZeroMQ](http://zeromq.org/) I'm experimenting with their [C4](http://rfc.zeromq.org/spec:22) process.

mvcisback, 2015-01-17:
@rickardlindberg I'm mainly interested in using it as a robotics platform. As such, the reactive part of FRP is very important to me. Ultimately, I'd like to see a DSL similar to Reactive Banana (#3)

nmarshall23, 2015-07-03:
@rickardlindberg I'm mainly interested in using the functional programming paradigm to write code in. I find that FP makes it far simpler to untangle program flow.

In addition FRP seems a nature fit for programming micro-controllers.

joefiorini, 2015-08-10:
@rickardlindberg I've been looking for a high-level language to make Arduino programming better, finding one that adopts FRP is a major, major bonus. I'm looking forward to trying this. I have varying interests in Arduino programming, but I want to support those interests with a language that is safe and allows me to express logic in a way that exposes itself to clear reasoning and functional principles like immutability.

AesaKamar, 2018-07-23:
I'm really excited about this project. Along with adopting more FP aligned languages like Haskell and Scala, I has started to feel like I was losing the ability to interact with embedded hardware in a satisfying and expressive way. But this perfect level of abstraction motivates me to start doing some hardware hacking again! 😄