frp-arduino/frp-arduino #15: Problem: PWM missing from API

mvcisback, 2015-01-17:


rickardlindberg, 2015-01-18:
I'm not entierly sure how PWM works on Arduino, but I imagine we would like an output like this:

    pwmPinX :: Output Byte

Then connect a stream of bytes to that output.

Such output should be possible to write with LLI (a really bad name, but stands for Low Level Instruction). For reference, see

    uart :: Output Byte

mvcisback, 2015-01-18:
[here](Bit-banging Pulse Width Modulation) is a nice primer on the 2 kinds of PWM supported by Arduino.

I think `Output Byte` byte would be **Bit-banging Pulse Width Modulation***. While sometimes this is desirable, there are many times when you want to use the built-in clocks.

The hardware clocks will work independent of cpu interrupts. While not currently used, interrupts might make for very natural Signal `Events`

rickardlindberg, 2015-01-19:
I thought PWM worked by saying how much time the signal should be high. So I thought a byte could represent 0=0% high, 255=100% high. But I'll have to read the AVR manual to see what the actual interface is.

rickardlindberg, 2015-01-19:
@mvcisback "here" link seems broken.

mvcisback, 2015-01-19:
@rickardlindberg you are correct, the Duty Cycle and the frequency define a PWM output.
So I imagine there would be a function:

```haskell
pwmOutput :: (Fractional a, Integral b) => a -> b -> Output PWM
pwmOutput duty freq = undefined
```

Also, looks like I wasn't paying attention when I C&P'd [here](http://arduino.cc/en/Tutorial/SecretsOfArduinoPWM). Its the arduino overview.