frp-arduino/frp-arduino #22: Issue handling process

rickardlindberg, 2015-01-21:
I'm trying to apply the C4 process to our issue handling. Specifically this part:

> The user or Contributor SHOULD write the issue by describing the problem they face or observe.
>
> The user or Contributor SHOULD seek consensus on the accuracy of their observation, and the value of solving the problem.
>
> Users SHALL NOT log feature requests, ideas, suggestions, or any solutions to problems that are not explicitly documented and provable.

Problems that we agree on can get the "agreed problem" label. Questions and
general discussions can get the "question" label.

mvcisback, 2015-01-21:
I'm fine with this. Could you clarify the not explicitly documented and provable part?

rickardlindberg, 2015-01-21:
I interpret it like this:

**explicitly documented:** The issue or commit does not state what the problem is.

**provable:** Not really sure about this one. It might be something like proposing a solution that claims to decrease compiled code size by 10% and there is no way of verififying, or proving, that the solution actually decreases code size. (Given that large compiled code size was a valid problem.)

There are some more clarifications in the ZeroMQ guide: http://zguide.zeromq.org/page:all#Development-Process

rickardlindberg, 2015-01-21:
I'm learning this process myself. I've never used it before. But I've read quite a lot about it. So this is a little experiment.

I think the most important part with the issue handling is focusing on *actual* problems. I think it's easy to run of and build a big solution, but the problem it solves might not be that important. That is what the process is designed to avoid.

I think the discussions we've had in the issues have all been interesting and useful. But in those cases where a real problem has not been identified, I propose to use the "question" label.

mvcisback, 2015-01-21:
Sounds good