frp-arduino/frp-arduino #37: Question: Could we use linear types to avoid duplicate resource allocation

mvcisback, 2015-02-06:
In the now resolved #1, the main problem is that after having used the resource it was still available for binding.

Linear types (based on [linear logic](http://en.wikipedia.org/wiki/Linear_logic)) encode this kind of invariant directly. [Here's](http://www.ioc.ee/~tarmo/tsem11/jeltsch1602-slides.pdf) a slide deck that overviews how to emulate these in Haskell.... but this is the kind of situation where Haskell's type system is abit wanting

Lastly as a bonus, languages like Rust have linear logic built-in for memory safety. That might make it a better target language (and there is even [work](https://github.com/jensnockert/dueboot) showing it working on Arduino)

rickardlindberg, 2015-02-07:
Encoding this in the types would be nice. As long as the types don't get too complicated.

Reminds me of this article: https://pragprog.com/magazines/2012-09/thinking-functionally-with-haskell

I'll have to read it again, but as I remember, it talked about the relationship between tests and types and further argued that types are a form of test.

rickardlindberg, 2015-02-07:
I didn't quite understand how linear types was used in the Rust example. Or maybe it wasn't? Maybe it was just that Rust code compiled to AVR?

mvcisback, 2015-02-07:
I suspect there are issues encoding this in Haskell.... but I'll need to look closer to see.

@rickardlindberg just as compiling to AVR. The linear types is built directly into the way it does memory management

mvcisback, 2015-02-08:
@rickardlindberg this is a great overview of linear types: [www.cs.ucsb.edu/~benh/290C_W15/papers/Linear Types Can Change the World.pdf](www.cs.ucsb.edu/~benh/290C_W15/papers/Linear Types Can Change the World.pdf)

rickardlindberg, 2015-02-09:
Interesting. I didn't understand all of it it the first pass. Maybe I'll read it once more another time :)