frp-arduino/frp-arduino #38: Question: Potentially add support for random sample rates

mvcisback, 2015-02-09:
Hey, I was wondering how hard it'd be to add support for random delays between samples.

The primary motivation is [compressive sensing](http://en.wikipedia.org/wiki/Compressed_sensing), where by assuming the signal is sparse, you can reconstruct signals given way fewer samples.

A good example is the single pixel camera from Rice University, which uses a similar technique to reconstruct a scene given only 1 pixel and random samples from the environment.

I think this matches well with the arduino since it has a relatively low possible sample rate (~9.5 kHz), but many processes (near field acoustics for example) require ~44.1kHz given uniform sampling. <-- I'm working on a process where I have to use expensive audio cards, but would really like to just use arduinos

rickardlindberg, 2015-02-09:
I'm not sure I understand what you mean. I'll write what I think you mean and then you can correct me if I'm wrong.

A sample sounds to me like the reading of an input. Specifically in our language, the `digitalRead` and `analogRead`.

When they are used today, a sample is read in every iteration. (The main while-loop is what iterates in the C code.)

What you want to do is to read only every other iteration? Or in general have a variable amount of delay between the reading of two samples?

If this is the case, I have two thoughts:

* This sounds like an optimization. Is there a problem reading a sample in every iteration? Does it become too slow in your use case? Or does the compressing sensing technique not work with too many samples?

* Modifying an input to only sample every n:th time called should not be that difficult to implement. Variable amount of delay might be harder depending on how it should vary.

mvcisback, 2015-02-09:
> A sample sounds to me like the reading of an input. Specifically in our language, the digitalRead and analogRead.

Yes

> What you want to do is to read only every other iteration? Or in general have a variable amount of delay between the reading of two samples?

This is a general variation in the amount of time between samples. I.E. the time between samples is a random variable between T_1 and T_2. For example consider a sine wave period `T`: `sin(2pi/T * t)`:

If the sampling is at regular intervals `[0, x, 2x, 3x, 4x, 5x ...]`, and rewriting `x` as a fraction of `T`:  `T = ax` we get

`[sin(2pi*0), sin(2pi*a), sin(2pi*2a), sin(2pi*3a)...]`

Note, that we can only have `x` unique samples. The point of sampling randomly is to get anywhere on the sin wave and thus more information per sample.