frp-arduino/frp-arduino #42: Project status

eallik, 2015-11-17:
Is the codebase currently suitable for production on GHC 7.10.x/7.10.2? The last commit was 7 months ago but the project looks appealing.

rickardlindberg, 2015-11-17:
The state of this project is more a proof-of-concept. There are many things missing.

Feel free to test it with GHC 7.10. Feel free to fork it and submit pull request if you find things that do not work.

rickardlindberg, 2015-11-17:
If you can't find an example that does something similar to what you want to do, then it probably can't be done. Have a look at the API as well.

I would not recommend this for production use.

I would recommend that you try it out though. Then, if it is good enough for your purposes, you might want to use it in production.

eallik, 2015-11-17:
I guess if I'm brave enough, I can just extend it? Or does that require full understanding of the denotative semantics of FRP?

rickardlindberg, 2015-11-17:
Yes, feel free to extend it. Just send a pull request and I will merge. Full understanding of FRP is not needed.

rickardlindberg, 2015-11-17:
If you have any questions, I will do my best to answer them.

eallik, 2015-11-17:
got it and thanks for the explanation!