frp-arduino/frp-arduino #43: How would one add new Arduino microcontrollers

ceigel, 2018-01-16:
I have an Arduino Micro. I see frp only has Arduino.Uno module. For another microcontroller would one need to add an Arduino.Micro module?

rickardlindberg, 2018-01-25:
Yes, that is probably what you have to do.

This project was just an experiment for me, and I'm no longer working on it. Feel free to experiment with the code.

jmwright, 2018-03-13:
I'm working on adding an [Arduino.Nano module](https://github.com/jmwright/frp-arduino/blob/master/src/Arduino/Nano.hs) to the project, and I'm capturing the process in [the docs](https://github.com/jmwright/frp-arduino/blob/master/doc/adding-boards.md). There are some board-specific options in the makefile, for example [here](https://github.com/frp-arduino/frp-arduino/blob/master/make#L31) and [here](https://github.com/frp-arduino/frp-arduino/blob/master/make#L35). The Nano uses a different baud rate than the Uno, and so the upload will fail without a manual change to the baud rate in the make file.

Is anyone opposed to inserting a BOARD parameter between the [EXAMPLE](https://github.com/frp-arduino/frp-arduino/blob/master/make#L5) and [TARGET](https://github.com/frp-arduino/frp-arduino/blob/master/make#L6) make file paramters? A conditional could then be used to select the appropriate parameters for Arduino-Makefile based on the board that is passed in.

jmwright, 2018-03-13:
I started working through this, and although I would prefer to leave `TARGET` as the last `make` parameter, it creates a situation where the board name can't easily be left out to default to an Uno. So I've put the BOARD parameter last.

rickardlindberg, 2018-03-13:
Your work looks good.

Another idea is to have BOARD be an environment variable that can be left out. To specify it:

    BOARD=Foo ./make

Or introduce proper argument parsing in the make script.

Just a few thoughts.

jmwright, 2018-03-13:
@rickardlindberg Thanks for the feedback. I like how the environment variable cleans things up a bit. It also allows the user to specify their board as a variable in .bashrc or .bash_profile so that it becomes their default board in place of the Uno. I've made that change.

jmwright, 2018-03-17:
@ceigel There is now documentation on how to add a board [here](https://github.com/frp-arduino/frp-arduino/blob/master/doc/adding-boards.md). I don't have access to an Arduino Micro like yours, but I can try to assist you in adding it if you want.

jmwright, 2018-03-23:
@ceigel Feel free to reopen this if needed.

closed 2018-03-23 by jmwright