frp-arduino/frp-arduino #46: API Documentation Missing

jmwright, 2018-03-12:
The [Readme](https://github.com/frp-arduino/frp-arduino#api) points users to the [frp-arduino Hackage page](http://hackage.haskell.org/package/frp-arduino) for the API reference. However, when you go to the the Hackage page and click on the [API link](http://hackage.haskell.org/package/frp-arduino-0.1.0.3/frp-arduino-0.1.0.3/src#api), you get a 404 (page not found) error.

jmwright, 2018-03-18:
@rickardlindberg on the Hackage page for the first two releases there are links for multiple sections of the API documentation, but in later releases those links have disappeared. It doesn't look like any of the links ever worked. Is there API documentation that got lost somewhere along the line? If you still have the API documentation files somewhere, I'll volunteer to get them integrated into the repo's docs if you send me the files.

rickardlindberg, 2018-03-18:
I can't find links in the readme that do not work. Which ones have you found?

I'm just on my phone now, so I just had a quick look.

jmwright, 2018-03-18:
For the API, the readme points to the Hackage page [here](http://hackage.haskell.org/package/frp-arduino). When you go to that page for version 0.1.0.3 and click on the API link in the index, you should get a 404 error. Also, there are no subsection links below the API link. However, if you go to the Hackage page for version 0.1.0.0 there are subsection links (Actions, Steam operations, etc) under API in the index, but they're all broken.

jmwright, 2018-03-19:
It looks like the API documentation was changed in this commit since it was in two places. It doesn't seem like the API documentation is in either place correctly now. https://github.com/frp-arduino/frp-arduino/commit/34e9ed750c574acb3f7ec81194a05328799391f3

jmwright, 2018-03-20:
Following on from my last post - I guess if there's a choice, I'd prefer to have the API docs hosted from the GitHub repo so it's easier to contribute to them, but I don't really understand the larger picture of the documentation and why Hackage was chosen for the API docs in the first place.

I'll be happy to do the work of fixing the docs, I just need some guidance on how they should be fixed. Here are a couple of screenshots that show how the API links in the index on Hackage have changed.

[image]

[image]

rickardlindberg, 2018-03-20:
The API doc is hosted on hackage because it is the standard location for Haskell API docs. It is like pypi in Python world.

The API doc is also automatically generated from the source code.

You can find the API doc on hackage now. Click the links to one of the modules at the top.

The table of contents links from the readme are broken though. Not sure why that is. But it looks like hackage has done changes to their website. Perhaps that did something to the table of contents links.

Did that make things clearer?

jmwright, 2018-03-20:
> Did that make things clearer?

Yes, thanks. I'm still learning about Haskell and its ecosystem, so I didn't think to look in the Modules section. In the next PR I do I'll probably add a short note to api.md to look in the Modules section on the Hackage page.

> But it looks like hackage has done changes to their website. Perhaps that did something to the table of contents links.

The images in the example sections are broken too. Instead of an image the following HTML text is what you'll see.
```
<p align="center"> <a href="http://youtu.be/UdIXmmp-6tw"> <img src="http://img.youtube.com/vi/UdIXmmp-6tw/0.jpg"> </a> </p> 
```
If simply uploading the latest code with the latest version of Cabal would fix the issues with the readme, I think we could justify a new release.

rickardlindberg, 2018-03-20:
I don't remember exactly how the page on Hackage is rendered. But it is for sure based on our README.md.

Perhaps this line does it? https://github.com/frp-arduino/frp-arduino/blob/master/frp-arduino.cabal#L11

I guess that Hackage does not allow markdown with embedded HTML.

I think my reasoning was that even though the rendering of the README is not perfect, it is better than nothing at all. So people only looking on Hackage will get some information from the README.

Making a new release will not solve the issues with the README, but it will make the API doc up to date. It might solve the broken table of contents links though.

jmwright, 2018-03-21:
> I guess that Hackage does not allow markdown with embedded HTML.

Looks that way. Hackage apparently uses the CheapSkate markdown renderer which doesn't support quite a few things, including embedded HTML.

I took a closer look at the Cabal project on Hackage, and it looks like they have different Readme.md files for GitHub versus Hackage. Once I'm a little more plugged into the Haskell/Hackage way of thinking maybe I'll take a swing at doing separate readme specifically for Hackage.

As far as releasing, [this](https://gist.github.com/tmcgilchrist/787532649671a9ece551) looks like a reasonable set of steps. The only one I don't really understand is 8.

jmwright, 2018-03-24:
@rickardlindberg Are you opposed to the idea of doing a release?

If we do one, I guess we'd maybe be at version 0.1.1.0 based on the Haskell package versioning policy listed [here](https://pvp.haskell.org/?rdfrom=https%3A%2F%2Fwiki.haskell.org%2Findex.php%3Ftitle%3DPackage_versioning_policy%26redirect%3Dno)

I can put a PR together using the steps from my last post if you are willing to do one. I think you'll have to be the one to push everything to Hackage. I'm assuming that takes a username/password combo. I think I can do everything else up to that step though.

rickardlindberg, 2018-03-24:
No, I am not opposed.

If you create an account on Hackage, I can give you privileges that allows you to upload a release.

jmwright, 2018-03-24:
@rickardlindberg I'm not sure if I have upload privileges quite yet, but my account has been created. The username is JeremyWright

rickardlindberg, 2018-03-25:
I've added you to the maintainer group. You should have privileges to upload a new version now. Let me know if it does not work.

jmwright, 2018-03-26:
The new release has been uploaded to [Hackage](http://hackage.haskell.org/package/frp-arduino), and as far as I can tell everything looks correct.

I also tagged a release [here on GitHub](https://github.com/frp-arduino/frp-arduino/releases/tag/0.1.1.0).

@rickardlindberg Thanks for the Hackage access and help getting up to speed.

closed 2018-03-26 by jmwright

rickardlindberg, 2018-03-26:
Good job. :+1: