frp-arduino/frp-arduino #47: test Fails Because of Missing Files in Dist

jmwright, 2018-03-15:
@rickardlindberg The tests are currently failing on my fork because the `src` files don't match what's in `dist/src`.
```
cabal file
  sdist
    contains all source code FAILED [1]
...
Failures:

  Spec.hs:78: 
  1) cabal file, sdist, contains all source code
       dist/frp-arduino-0.1.0.3> diff -r ../../src src > /home/jwright/Downloads/repos/frp-arduino/test_report.txt 2>&1
       
       Only in ../../src/Arduino: Nano.hs
```

Is this to force a release each time a significant change is made to the codebase? If so, could you explain the procedure for doing the release properly?

rickardlindberg, 2018-03-15:
I believe it is to ensure that when a release is made, all files are present.

The solution for you is probably to add your module to the cabal file.

rickardlindberg, 2018-03-15:
This file: https://github.com/frp-arduino/frp-arduino/blob/master/frp-arduino.cabal

jmwright, 2018-03-15:
That fixed it, thanks.

closed 2018-03-15 by jmwright