frp-arduino/frp-arduino #51: Not able to run make succesfully

chrissound, 2019-04-17:
BOARD=Uno make Blink upload      
make: *** No rule to make target 'Blink'.  Stop.

<details>
  <summary>BOARD=Uno make -np Blink upload</summary>
 # GNU Make 4.2.1
# Built for x86_64-pc-linux-gnu
# Copyright (C) 1988-2016 Free Software Foundation, Inc.
# License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
# This is free software: you are free to change and redistribute it.
# There is NO WARRANTY, to the extent permitted by law.

# Make data base, printed on Wed Apr 17 22:23:04 2019

# Variables

# automatic
<D = $(patsubst %/,%,$(dir $<))
# automatic
?F = $(notdir $?)
# environment
DESKTOP_SESSION = /nix/store/19k6msw4lcfszk7iak0h9ba3navqzffv-desktops/share/xsessions/xterm
# default
.SHELLFLAGS := -c
# environment
__NIXOS_SET_ENVIRONMENT_DONE = 1
# environment
XDG_SESSION_CLASS = user
# environment
XDG_SESSION_TYPE = x11
# default
CWEAVE = cweave
# automatic
?D = $(patsubst %/,%,$(dir $?))
# automatic
@D = $(patsubst %/,%,$(dir $@))
# environment
XAUTHORITY = /home/chris/.Xauthority
# automatic
@F = $(notdir $@)
# default
MAKE_VERSION := 4.2.1
# makefile
CURDIR := /home/chris/fromLaptopt/usbflash/Haskell/AstatineAlphaAvocado/frp-arduino
# makefile
SHELL = /bin/sh
# default
RM = rm -f
# environment
NIXPKGS_CONFIG = /etc/nix/nixpkgs-config.nix
# environment
VTE_VERSION = 5202
# environment
_ = /run/current-system/sw/bin/make
# default
PREPROCESS.F = $(FC) $(FFLAGS) $(CPPFLAGS) $(TARGET_ARCH) -F
# default
LINK.m = $(OBJC) $(OBJCFLAGS) $(CPPFLAGS) $(LDFLAGS) $(TARGET_ARCH)
# environment
XDG_CONFIG_DIRS = /home/chris/.nix-profile/etc/xdg:/nix/var/nix/profiles/default/etc/xdg:/run/current-system/sw/etc/xdg:/etc/profiles/per-user/chris/etc/xdg
# default
OUTPUT_OPTION = -o $@
# default
COMPILE.cpp = $(COMPILE.cc)
# makefile
MAKEFILE_LIST := 
# 'override' directive
GNUMAKEFLAGS := 
# default
LINK.p = $(PC) $(PFLAGS) $(CPPFLAGS) $(LDFLAGS) $(TARGET_ARCH)
[... 1234 lines cut ...]
	$(LINK.f) $^ $(LOADLIBES) $(LDLIBS) -o $@

# Not a target:
.info:
#  Builtin rule
#  Implicit rule search has not been done.
#  Modification time never checked.
#  File has not been updated.

# files hash-table stats:
# Load=77/1024=8%, Rehash=0, Collisions=522/1941=27%
# VPATH Search Paths

# No 'vpath' search paths.

# No general ('VPATH' variable) search path.

# strcache buffers: 1 (0) / strings = 618 / storage = 8026 B / avg = 12 B
# current buf: size = 8162 B / used = 8026 B / count = 618 / avg = 12 B

# strcache performance: lookups = 1020 / hit rate = 39%
# hash-table stats:
# Load=618/8192=8%, Rehash=0, Collisions=262/1020=26%
# Finished Make data base on Wed Apr 17 22:23:04 2019
</details>

jmwright, 2019-04-18:
@chrissound It's been a long time, but I think the Uno should be selected by default. Try running it without the `BOARD=Uno` variable declaration. I think you only need to include that if you're using a Nano.

chrissound, 2019-04-19:
No luck with that either, I would assume it's related to the new changes in the related repo `sudar/Arduino-Makefile`.

chrissound, 2019-04-19:
I don't think running through this with `make` is strictly required, but it seems I'm able to just use `arduino --upload example.ino --port /dev/ttyUSB*` on the compiled c file (renamed to example.ino).

Haven't got it to blink a LED yet but probably something else needs configuring (I'm actually using a mega).

jmwright, 2019-04-19:
To get the blink example to work when uploading manually, you will probably have to add Mega support. There's information on adding board support [here](https://github.com/frp-arduino/frp-arduino/blob/master/doc/adding-boards.md). After you've added the board, I think you'll want to change any mention of `Arduino.Uno` to `Arduino.Mega` (or whatever you name the new board) in the blink example. That will hopefully fix any port mapping and memory size issues that are keeping the blink example from working. I'm telling this all from memory though, so keep that in mind.

When I get a chance I'll try upgrading my version of Arduino-Makefile to see what's going on. I'm still running on the old version I used when I contributed the changes that enabled the BOARD variable.

jmwright, 2019-05-12:
@chrissound I cloned the latest version of Arduino-Makefile (from their GitHub master) into the root frp-arduino directory, and it works fine for me. Did you by chance install the arduino-mk package on either Linux or Mac? If so, I think you'll need a copy of the Arduino-Makefile directory in the root directory of frp-arduino to make things work.