frp-arduino/frp-arduino #55: Support for stack

miladhub, 2019-10-22:
Hi,
I don't have GHC globally installed, but I do have Stack, so in my local environment (Mac), I made this modification and everything works:
```
$ git diff
diff --git a/make b/make
index b74fec1..8c07a1b 100755
--- a/make
+++ b/make
@@ -35,7 +35,7 @@ then
     rm -rf $OUTPUT_DIR
 else
     mkdir -p $OUTPUT_DIR
-    ghc \
+    stack exec -- ghc \
         --make \
         -Werror \
         -fwarn-unused-imports \
```
It would be nice if this were auto-detected.

miladhub, 2019-10-26:
My intent is simply to use stack here. For the benefit of reproducibility, I've created a project here: https://github.com/miladhub/myarduino copy-pasting `Blink.hs` into `Main.hs` and using the resolver version that should match the ghc version that you tested the project with.

Command `stack build` yields:

    /private/var/folders/2l/76dm15b56vq3vdd1g16886w00000gn/T/stack4252/frp-arduino-0.1.1.0/src/Arduino/Internal/CodeGen/C.hs:333:17: error:
        • No instance for (Control.Monad.Fail.MonadFail
                             Data.Functor.Identity.Identity)
            arising from a do statement
            with the failable pattern ‘(Value x cType _ _)’
        • In a stmt of a 'do' block: (Value x cType _ _) <- genLLI value
          In the expression:
            do (Value x cType _ _) <- genLLI value
               block ("if (" ++ x ++ ") {")
                 $ do line (register ++ " |= (1 << " ++ bit ++ ");")
               block "} else {"
                 $ do line (register ++ " &= ~(1 << " ++ bit ++ ");")
               line "}"
          In a case alternative:
              _ -> do (Value x cType _ _) <- genLLI value
                      block ("if (" ++ x ++ ") {")
                        $ do line (register ++ " |= (1 << " ++ bit ++ ");")
                      block "} else {"
                        $ do line (register ++ " &= ~(1 << " ++ bit ++ ");")
                      ....
        |
    333 |                 (Value x cType _ _) <- genLLI value
        |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

What am I doing wrong?

jmwright, 2019-10-26:
I wonder if this is related.

https://github.com/spell-music/csound-catalog/issues/4

miladhub, 2019-11-03:
Just for the record, this works:

    stack init
    stack build

so I don't know why using this project as a dependency doesn't work. Right now I'm using it as an additional project after copy-pasting it into my project:

    mkdir frp-arduino
    cp -r /path/to/frp-arduino fro-arduino

and modify `make` follows:

```
$ git diff
diff --git a/make b/make
index b74fec1..5a2d848 100755
--- a/make
+++ b/make
@@ -10,9 +10,10 @@ if [ "$#" -lt 1 ]; then
                exit 1
 fi
 
+FRPDIR=$(dirname "$0")
 SOURCE=$1
 TARGET=$2
-OUTPUT_DIR=build-output/$(basename "$SOURCE")
+OUTPUT_DIR=$FRPDIR/build-output/$(basename "$SOURCE")
 BASENAME=$(basename "$SOURCE")
 
 # We do not want the file extension at this time
@@ -35,7 +36,7 @@ then
     rm -rf $OUTPUT_DIR
 else
     mkdir -p $OUTPUT_DIR
-    ghc \
+    stack exec -- ghc \
         --make \
         -Werror \
         -fwarn-unused-imports \
```

1000000000, 2019-12-29:
I found it was possible to fix the MonadFail compile error by using an older version of GHC which I did by setting my stack resolver to `lts-12.26`. I suspect the fact that manually importing it as a package works is because the frp-arduino version here on Github (which is the one you are using) is newer than the one on Hackage and has had that particular compiler error (which caused by changes in newer versions of GHC) fixed.